microsoft/mu_basecore #289: [Bug]: Run BuildEnv Script bring a Error in Line 233

Ratatouilledummie, 2023-02-28:
### Is there an existing issue for this?

- [X] I have searched existing issues

### Current Behavior

Run BuildEnv script in Basetools after make basetools, bring a error of unary operator expected.
Output:
Loading previous configuration from /home/user/mu_basecore/BaseTools/Conf/BuildEnv.sh
Using EDK2 in-source Basetools
WORKSPACE: /home/user/mu_basecore/BaseTools
EDK_TOOLS_PATH: /home/user/mu_basecore/BaseTools
CONF_PATH: /home/user/mu_basecore/BaseTools/Conf
./BuildEnv: line 223: [: !=: unary operator expected
Copying $EDK_TOOLS_PATH/Conf/build_rule.template
     to /home/user/mu_basecore/BaseTools/Conf/build_rule.txt
./BuildEnv: line 223: [: !=: unary operator expected
Copying $EDK_TOOLS_PATH/Conf/tools_def.template
     to /home/user/mu_basecore/BaseTools/Conf/tools_def.txt
./BuildEnv: line 223: [: !=: unary operator expected
Copying $EDK_TOOLS_PATH/Conf/target.template
     to /home/user/mu_basecore/BaseTools/Conf

### Expected Behavior

Set the Environment for use BinWrappers 

### Steps To Reproduce

1. Build BaseTools manually using make.
2. Run BuildEnv script inside BaseTools

### Build Environment

```markdown
- OS(s): Ubuntu 18.04, 20.04, 22.04 can be afected more systems, i have tried in some docker containers
- Tool Chain(s): GCC-5
- Targets Impacted: NO-TARGET
```

### Version Information

```text
Release: 202208
```

### Urgency

Low

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

No maintainer feedback needed

spbrogan, 2023-03-02:
In project mu we don't actually use/maintain that method of building the basetools.  I would suggest using https://github.com/microsoft/mu_basecore/blob/release/202208/BaseTools/Edk2ToolsBuild.py
as that is supported for both linux or windows builds and is tested regularly.  

Any concerns?

Ratatouilledummie, 2023-03-02:
> In project mu we don't actually use/maintain that method of building the basetools. I would suggest using https://github.com/microsoft/mu_basecore/blob/release/202208/BaseTools/Edk2ToolsBuild.py as that is supported for both linux or windows builds and is tested regularly.
> 
> Any concerns?

Thanks, sorry for my confusion, thanks

makubacki, 2023-03-03:
Thank you for reaching out to us. I'm going to close this issue based on the discussion.

closed 2023-03-03 by makubacki