microsoft/mu_basecore #307: VS2022 ARM64 build is broken due a linker error in DxeCore

artgup, 2023-03-07:
### Is there an existing issue for this?

- [X] I have searched existing issues

### Current Behavior

Compile the Hyper-V UEFI code using the latest VS2022. It gives a linker error

INFO -   "C:\Program Files\Microsoft Visual Studio\2022\Enterprise\VC\Tools\MSVC\14.35.32215\bin\Hostx86\arm64\lib.exe" /NOLOGO /LTCG /OUT:d:\a\1\s\Build\MsvmAARCH64\RELEASE_VS2022\AARCH64\MsGraphicsPkg\MsUiTheme\Pei\MsUiThemePpi\OUTPUT\MsUiThemePpi.lib @d:\a\1\s\Build\MsvmAARCH64\RELEASE_VS2022\AARCH64\MsGraphicsPkg\MsUiTheme\Pei\MsUiThemePpi\OUTPUT\object_files.lst
INFO - DxeCore.lib(HeapGuard.obj) : error LNK2001: unresolved external symbol memset
INFO - d:\a\1\s\Build\MsvmAARCH64\RELEASE_VS2022\AARCH64\MdeModulePkg\Core\Dxe\DxeMain\DEBUG\DxeCore.dll : fatal error LNK1120: 1 unresolved externals

### Expected Behavior

Compiling and linking should work.

### Steps To Reproduce

Compile Mu_Basecore for ARM64 with VS2022

### Build Environment

```markdown
- OS(s):Windows
- Tool Chain(s):VS2022
- Targets Impacted: RELEASE
```

### Version Information

```text
Release/202208
```

### Urgency

Medium

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

No maintainer feedback needed

makubacki, 2023-03-07:
Current workaround is to link MdePkg/Library/CompilerIntrinsicsLib/ArmCompilerIntrinsicsLib.inf against the module which provides `memset`. Further investigation needs to be done to determine what code in HeapGuard led to the issue.

Similar to #291. Occurred on the same VS version update.

makubacki, 2023-04-18:
I could reproduce the issue on VS2022 17.5.1. There I saw `memset` being used in the following locations:

- MakeTable() in BaseUefiDecompressionLib.c
- SetGuardedMemoryBits() in HeapGuard.c
- ClearGuardedMemoryBits() in HeapGuard.c

I cannot reproduce the issue with 17.5.3 so I am closing this issue. Please reopen if it reproduces on 17.5.3+.

closed 2023-04-18 by makubacki