microsoft/mu_basecore #371: [Feature]:  xdrlib deprecation and removal in python 3.13

apop5, 2023-04-24:
### Feature Overview

Python 3.11.3 has started to report warnings in regards to the xdrlib.

`ini
MU_BASECORE\BaseTools\Scripts\BinToPcd.py:16: DeprecationWarning: 'xdrlib' is deprecated and slated for removal in Python 3.13`

This needs to be addressed prior to 3.13 release, and we need to be aware that this will may cause issues supporting older code-bases on newer python tools.

### Solution Overview

BinToPcd.py is the only xdrlib usage, currently. 

### Urgency

Low

### Are you going to implement the feature request?

Someone else needs to implement the feature

### Do you need maintainer feedback?

No maintainer feedback needed

makubacki, 2023-04-24:
Note that I filed a TianoCore bug on 12-09-2022 and Mike Kinney took ownership to fix it.

[Bug 4188 - BinToPcd.py - xdrlib is deprecated in Python 3.11; removal in Python 3.13](https://bugzilla.tianocore.org/show_bug.cgi?id=4188)

Javagedes, 2023-05-02:
I will take note to monitor this and cherry-pick it when Bug 4188 is resolved.

PR #473 (2023-06-28): BaseTools: BinToPcd: replace xdrlib functionality
## Description

Replaces xdrlib with a custom packing function mimicking current xdrlib usage as xdrlib is being deprecated in Python 3.13 with no clear library replacement.

closes #371

- [ ] Impacts functionality?
  - **Functionality** - Does the change ultimately impact how firmware functions?
  - Examples: Add a new library, publish a new PPI, update an algorithm, ...
- [ ] Impacts security?
  - **Security** - Does the change have a direct security impact on an application,
    flow, or firmware?
  - Examples: Crypto algorithm change, buff [... cut]
commit 3e1cb8dc (2023-06-28): BaseTools: BinToPcd: replace xdrlib functionality (#473)

## Description

Replaces xdrlib with a custom packing function mimicking current xdrlib
usage as xdrlib is being deprecated in Python 3.13 with no clear library
replacement.

closes #371

- [ ] Impacts functionality?
- **Functionality** - Does the change ultimately impact how firmware
functions?
- Examples: Add a new library, publish a new PPI, update an algorithm,
...
- [ ] Impacts security?
- **Security** - Does the change have a direct security impact on an
application,
    flow, or firmware?
  - Examples: Crypto algorithm change, bu [... cut]
closed 2023-06-28 by Javagedes