microsoft/mu_basecore #386: [Bug]: AARCH64 LD Can Misinterpret Multiple common-page-size Parameters

os-d, 2023-05-02:
### Is there an existing issue for this?

- [X] I have searched existing issues

### Current Behavior

It has been observed that on aarch64 package builds, GenFw can fail at this line: https://github.com/microsoft/mu_basecore/blob/1bd4b3bfd657ad5a3940d072be6540f7c9c23c5d/BaseTools/Source/C/GenFw/Elf64Convert.c#L1540.

What appears to be happening is that ld is interpreting the multiple common-page-size parameters passed to it in a different order, sometimes at 0x20, sometimes at 0x1000, which breaks GenFw's expectation of alignment. It does not appear to be valid to pass common-page-size to ld multiple times.

There are complicating factors: the same toolchain will work on the same code, then it will break the next day, with no code changed. Manually editing tools_def.txt (or tools_def.template and regenerating tools_def.txt) to set all common-page-size calls for AARCH64 GCC to 0x1000 causes the issue to go away, however, on a system that is not experiencing the issue, setting all the common-page-size parameters to 0x20 does not repro the issue.

The issue does repro occasionally across multiple machines, both local dev boxes of different devs (and on different packages) and on pipelines. So far, this has only been observed in non-public code; I am attempting to get a public repro, however, with the unpredictability of getting a repro, I have been unsuccessful so far.

At the very least, not passing common-page-size multiple times would seem to be a step in the right direction, although further investigation here seems warranted.

### Expected Behavior

I expect GenFw to not fail and properly be able to convert ELF files to PE/COFF.

### Steps To Reproduce

Unfortunately, this is not easily reproducible, but the signature is clear.

### Build Environment

```markdown
- OS(s): Linux
- Tool Chain(s): GCC
- Targets Impacted: AARCH64
```

### Version Information

```text
Version v2022080001.4.2
```

### Urgency

High

### Are you going to fix this?

I will fix it

### Do you need maintainer feedback?

No maintainer feedback needed

os-d, 2023-05-04:
Commit: https://github.com/microsoft/mu_basecore/commit/1bd4b3bfd657ad5a3940d072be6540f7c9c23c5d, was introduced as a potential fix to this issue, however, it was intended to prevent issues from being seen while the issue was root caused. However, the commit prevented the issue from being seen and standalone repros were proving difficult to create. At the same time, this commit expanded the size of PEI modules, which can break platforms expecting smaller sizes, such as in mu_tiano_platforms: https://github.com/microsoft/mu_tiano_platforms/pull/535.

A [PR](https://github.com/microsoft/mu_basecore/pull/391) was put up to revert the above commit, so that random repros would be generated and the issue could be root caused without requiring PEI modules to greatly increase in size.

makubacki, 2023-06-23:
Lowering the urgency given the time the issue has been open (~1.5 months).

closed 2023-08-16 by github-actions[bot]