microsoft/mu_basecore #395: [Bug]: Access Denied Seen on Some Windows Systems with BaseTools bin 2022.08.4_1

makubacki, 2023-05-08:
### Is there an existing issue for this?

- [X] I have searched existing issues

### Current Behavior

Builds have failed on local systems and build agents after updating to 2022.08.4_1 by encountering a "Access Denied" error.

For example, the error was raised when running `GenSec -s EFI_SECTION_PE32 -o <path>/.../<guid>1.pe32`

The issue can be worked around on affected systems by excluding the source directory in Windows Defender.

### Expected Behavior

A regression in Windows Defender compatibility is not present or readily understood.

### Steps To Reproduce

1. Build a platform based on mu_basecore v2022080002.0.1
2. See if the error repros when running `GenSec` on the system
  - > Note: It has been reported to not repro on all systems

### Build Environment

```markdown
N/A - Basetools bin release
```

### Version Information

```text
Tag: v2022080002.0.1
```

### Urgency

Medium

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

Maintainer feedback requested

kuqin12, 2023-05-16:
Do not see repro on those systems used to report failure. Will keep this open for a bit longer to see if the BaseTools are added to allow list.

makubacki, 2023-06-23:
Closing since the issue no longer repos

closed 2023-06-23 by makubacki