microsoft/mu_basecore #409: [Bug]: ARM Platforms Do Not Sync RP/RO/XP Bits to GCD Initially Leading to Access Flag Fault

os-d, 2023-05-22:
### Is there an existing issue for this?

- [X] I have searched existing issues

### Current Behavior

Recently, EDK2 updated so that ARM would use TT_AF (the access flag in the page table) to represent EFI_MEMORY_RP (read protected memory). This enforces an access flag fault when software
attempts to read a page without the access flag set. This is one option allowed by the Arm ARM; previously EDK2 was using TT_AF in the other allowed way, which used TT_AF as an accessed bit,
i.e. a page would not have this bit set unless the page had been accessed.

Project Mu added a change to properly sync the GCD with the page on ARM, as is done on the x86 side, but was never added to the Arm side due to an oversight upstream. This allowed all memory
regions to have the capability of setting EFI_MEMORY_RP as an attribute in the GCD, which they did not have previously. When Arm's CpuDxe syncs with the GCD, it attempts to set all ranges of
System Memory as EFI_MEMORY_RP by default, because that is how they start out in the page table (this is a change in behavior on the upstream side, previously the access flag is set for all
ranges). Because the Capabilities of each GCD range now supports EFI_MEMORY_RP getting set, it would also get marked in the GCD as EFI_MEMORY_RP. Previously, the Arm implementation would
silently drop the EFI_MEMORY_RP bit being passed in as an attribute to set because it was not in the capability set of the GCD region.

When the USB stack runs on platforms that use NonDiscoverablePciDeviceIo to manage the MMIO allocations, NonCoherentPciIoAllocateBuffer gets a new page, saves off the existing GCD descriptor,
which contains EFI_MEMORY_RP for this range, updates the region in the GCD (which in turn updates the page table) to not include EFI_MEMORY_RP (implicitly when setting a cacheability attribute).
When the memory is freed through NonCoherentPciIoFreeBuffer, the saved off GCD descriptor is fetched and it resets the memory range to have the old memory attributes, which includes EFI_MEMORY_RP.

When freeing a memory pool, the core code attempts to read the page (to set some metadata about the status of the pool, return it to the free list, etc.), which then triggers the access flag fault.

This did not fail on debug builds, because in debug builds we set the CLEAR_MEMORY_ENABLED bit, which, on every conversion of memory type (say in allocating a page) and on every pool allocate
and free, causes us to call CLEAR_MEMORY_ENABLED(), which clears the EFI_MEMORY_RP bit (setting the TT_AF flag), so that future accesses of this do not get an access flag fault. We should
probably be unsetting the RP bit on every free, as the core code will fail when attempting to give it out.

The issue is in Arm's MMU mgmt code, specifically GetNextEntryAttribute where it has EntryAttribute = Entry & TT_ATTR_INDX_MASK, which evaluates to 00011100, but TT_AF is BIT10 (1000000000) so it always gets masked off
which in the old implementation worked, because we were only using it to tell if a page had been accessed, so the GCD didn't need to know about it. This should be TT_ATTRIBUTE_MASK so that we pick up TT_AF as well
as the RO and XP equivalent bits in the page table entry that we were missing on initial GCD/page table syncing prior to this.

### Expected Behavior

We should not hit an access flag fault when freeing a page.

### Steps To Reproduce

Run an ARM platform using a nondiscoverable PCI device with PcdDebugPropertyMask with the CLEAR_MEMORY_ENABLED bit set to 0 with the latest mu_basecore release.

### Build Environment

```markdown
- OS(s): Ubuntu 20.04
- Tool Chain(s): GCC5
- Targets Impacted: RELEASE
```

### Version Information

```text
Tag: v2022080002.0.1
```

### Urgency

High

### Are you going to fix this?

I will fix it

### Do you need maintainer feedback?

No maintainer feedback needed

os-d, 2023-06-07:
The top level issue is resolved through the linked PR, but the FreePool code still has the possibility of failure when freeing an EFI_MEMORY_RP region. Leaving this open to track that work.

makubacki, 2023-06-23:
Lowering the urgency given the linked PR

closed 2023-08-14 by github-actions[bot]