microsoft/mu_basecore #503: [Feature]: Provide a hook API when stack overflow occur with stack protection enabled

MarcChen46, 2023-07-27:
### Feature Overview

**[Background]**
Currently we have EFI_CPU_ARCH_PROTOCOL.RegisterInterruptHandler() for DXE_DRIVER and EFI_SMM_CPU_SERVICE_PROTOCOL.RegisterExceptionHandler() for DXE_SMM_DRIVER to register EXCEPT_IA32_STACK_FAULT type that can perform some customized actions when stack overflow occurred with stack protection enabled, but it has a dependency that the customized code can only get performed after these protocols get installed, it is the gap we want to resolve.

**[Feature request]**
When stack overflow occurred in any protected modules types(ex: DXE_DRIVER, DXE_CORE, SMM_CORE, DXE_SMM_DRIVER...), it should provide a hook API or mechanism that allow platform to add customized code (ex: report the error to somewhere as telemetry), the customized should not have dependency on protocols.

### Solution Overview

Below is just a proposal for reference, it is ok to have any other kind of methods to achieve it.
**[Proposal]**
Implement a MU version of BaseStackCheckLib instance and implement below code in BaseStackCheckMsft.c, use MSFT VS compiler to compile this C code that has an customized hook API inside __report_gsfailure.

```
NORETURN VOID __cdecl
__report_gsfailure (
  void
  )
{
  DEBUG ((DEBUG_ERROR, "(__report_gsfailure) STACK PROTECTION : Violation detected\n"));

  CustomizedHook ();  // API provided by another new added library class
  CpuDeadLoop ();
}
```

### Urgency

Low

### Are you going to implement the feature request?

Someone else needs to implement the feature

### Do you need maintainer feedback?

Maintainer feedback requested

MarcChen46, 2023-09-18:
@TaylorBeebe, sorry, I did not notice the issue will be closed automatically, may I know if there is any plan for this feature request? can we re-open this issue?

TaylorBeebe, 2023-09-18:
> @TaylorBeebe, sorry, I did not notice the issue would be closed automatically, may I know if there is any plan for this feature request? can we re-open this issue?

This feature request is in progress -- reopened the issue

TaylorBeebe, 2023-11-03:
@MarcChen46
This feature will be completed soon. I posted a draft PR which will be cleaned up over the coming days. I'll ping this issue when the PR is out of draft status. https://github.com/microsoft/mu_basecore/pull/610

TaylorBeebe, 2023-11-03:
PR is out of draft status.

TaylorBeebe, 2023-11-22:
Feature completed

closed 2023-11-22 by TaylorBeebe