microsoft/mu_basecore #506: [Feature]: Improve performance for NvmExpressDxe driver

athenaz2, 2023-08-01:
### Feature Overview

Current driver binding start function for NvmExpressDxe driver is being called multiple times and moving the stall to after the status check improves its performance.

### Solution Overview

Moving stall function from before the status check to after the status check.

### Urgency

Low

### Are you going to implement the feature request?

I will implement the feature

### Do you need maintainer feedback?

Maintainer feedback requested

makubacki, 2023-08-03:
Thank you for creating PR #505. As noted there, @apop5 will help send this patch for you to the edk2 mailing list. Therefore, I'm assigning the issue to @apop5. 

For reference, the patch is readily available [here](https://patch-diff.githubusercontent.com/raw/microsoft/mu_basecore/pull/505.patch).

apop5, 2023-09-18:
Pending.

closed 2023-11-10 by github-actions[bot]