microsoft/mu_basecore #695: [Feature]: Stack Protect for PEIM in IA32

MarcChen46, 2024-01-25:
### Feature Overview

Current stack protect feature has not support PEIM in IA32 yet, it is critical to get the stack protect in both PEI and DXE phase, please plan to implement the stack protect for PEIM in IA32 and it would be great if it can be completed in early February.

### Solution Overview

Provide the same stack protect feature as DXE phase by StackCheckLib.

### Urgency

High

### Are you going to implement the feature request?

Someone else needs to implement the feature

### Do you need maintainer feedback?

Maintainer feedback requested

### Anything else?

@apop5 @TaylorBeebe

TaylorBeebe, 2024-02-02:
This is being worked on. A PR is up here: https://github.com/microsoft/mu_basecore/pull/714

This PR requires some CI issues to be resolved:

1. The build-time stack cookie logic requires edits to base tools. In CI, we use the EDK2 basetools from Tianocore. The EDK2 basetools can either be dropped from pip-requirements, or the setting in CISettings should be updated to specify that the local basetools should be used.

2. Sometimes, more than one stack cookie library can be included for a module. This is inconsistent and only seems to happen in CI which indicates there may be a bug in basetools. This is being investigated.

TaylorBeebe, 2024-03-14:
This feature has been committed

closed 2024-03-14 by TaylorBeebe