microsoft/mu_basecore #1208: [Bug]: subhook submodule is removed from upstream Zeex/subhook.git

aliibrahim8800, 2024-11-04:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Current Behavior

The upstream repository has been removed https://github.com/Zeex/subhook.git

All of Zeex repositories have been removed.

### Expected Behavior

Maintaining subhook internally.

### Steps To Reproduce

1. Recursively clone mu_basecore. 
2. Will fail the clone process with missing Zeex/subhook

### Build Environment

- OS(s): Windows 11 and Ubuntu 22.04
- Tool Chain(s): GCC5 and VS2022
- Targets Impacted: All

### Version Information

N/A

### Urgency

High

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

Maintainer feedback requested

apop5, 2024-11-04:
https://github.com/tianocore/edk2/issues/6398

This is coming from upstream.

Please follow the topic there since this affects all edk2 consumers.

simon-zheng-amd, 2024-11-04:
Encountered same issue here, an possible alternetive replacement is https://github.com/BangL/subhook for https://github.com/Zeex/subhook

makubacki, 2024-11-05:
Closed with:

- dev/202405: [[CHERRY-PICK] UnitTestFrameworkPkg: Use TianoCore mirror of subhook submodule by makubacki · Pull Request #1209 · microsoft/mu_basecore](https://github.com/microsoft/mu_basecore/pull/1209)
- release/202311: [[CHERRY-PICK] UnitTestFrameworkPkg: Use TianoCore mirror of subhook s… by makubacki · Pull Request #1210 · microsoft/mu_basecore](https://github.com/microsoft/mu_basecore/pull/1210)
- release/202302: [[CHERRY-PICK] UnitTestFrameworkPkg: Use TianoCore mirror of subhook submodule by makubacki · Pull Request #1211 · microsoft/mu_basecore](https://github.com/microsoft/mu_basecore/pull/1211)
- release/202208: [[CHERRY-PICK] UnitTestFrameworkPkg: Use TianoCore mirror of subhook s… by makubacki · Pull Request #1212 · microsoft/mu_basecore](https://github.com/microsoft/mu_basecore/pull/1212)

closed 2024-11-05 by makubacki